devicetree-org/lopper #2: bug in parsing .dts files

bentheredonethat, 2020-06-08:
Given certain values in the reg property of node, value not parsed properly.

sample input device-trees/system-device-tree.dts attached. output for the node

        channel0vdev0buffer: channel0vdev0buffer@3ed48000 {
                no-map;
                #address-cells = <2>;
                #size-cells = <2>;
                reg = <0x3ed48000 0x100000>;
                compatible = "openamp,xlnx,mem-carveout";
        };

should print the register value as 0x3ed48000 0x100000
but when parsed via 
+    for k,v in sdt.tree.__nodes__.items():
+        if "channel0" in k: #and "openamp,xlnx,mem-carveout" in v["compatible"]:
+            print(v['reg'])

output is reg = ">Ԁ","","","";

[system-device-tree.dts.txt](https://github.com/devicetree-org/lopper/files/4747596/system-device-tree.dts.txt)

zeddii, 2020-06-08:
I'll have a look ASAP. It's likely just the decode that is busted, there are a lot of corner cases with libfdt. Assume that you are getting the right output, and I'll look into a fix + regression run.

zeddii, 2020-06-09:
Author: Bruce Ashfield <bruce.ashfield@xilinx.com>
Date:   Tue Jun 9 15:43:45 2020 -0400

    lopper/fdt: add a secondary string validity test
    
    Some properties were being mistaken for strings. Add a second
    test for strings to confirm when the guess routines think
    something is a string.
    
    When both vote yet, we are much more sure that we actually have
    a string.
    
    Signed-off-by: Bruce Ashfield <bruce.ashfield@xilinx.com>

:100644 100644 1252330... 778d85c... M  lopper_fdt.py

zeddii, 2020-06-09:
This resolves the decoding in my sanity tests (I've added that property to lopper_sanity as well).

closed 2020-06-09 by zeddii