devicetree-org/lopper #30: setup lopper as a pypi package

galak, 2021-03-11:
Add whatever needs to be do such that lopper is available via pypi

galak, 2021-03-11:
@mbolivar-nordic this might be something you can help @zeddii with.

mbolivar-nordic, 2021-03-15:
@zeddii are there any:

- .py files in the repo that you do **not** want on PyPI?
- non-.py files in the repo that you **do** want on PyPI?

That's enough for me to send a PR with the packaging files. I'll make sure `pip3 install lopper` installs a `lopper` executable, and I'd also recommend making `python3 -m lopper` have the same effect (running `lopper`).

mbolivar-nordic, 2021-03-15:
Oh, and if you can tell me the following metadata:

- author name(s)
- email
- homepage

zeddii, 2021-03-15:
> @zeddii are there any:
>
> * .py files in the repo that you do **not** want on PyPI?
> * non-.py files in the repo that you **do** want on PyPI?
>
> That's enough for me to send a PR with the packaging [...]

Anything (python or not) that are part of the current git repo are find for pypi, even if they aren't directly used.

I haven't done an audit of files that are customarily used, but I'd like to keep it simple and just go with that is in git.

zeddii, 2021-03-15:
> Oh, and if you can tell me the following metadata:
> 
> * author name(s)
> * email
> * homepage

Author Name:  Bruce Ashfield
Email: bruce.ashfield@gmail.com

As for homepage, we are just using the github project as the landing page for everything, with references to that from other places. So we might as well just use that.

zeddii, 2021-12-09:
the master branch is now pypi enabled!

closed 2021-12-09 by zeddii