devicetree-org/lopper #32: system devicetree should support describing a board w/multiple different SoCs

galak, 2021-03-18:
An example from the MCU world is the [96board Carbon](https://www.96boards.org/product/carbon/) in which we have a ST STM32F401 and Nordic NRF 51822.  The NRF is acting as a bluetooth radio.

We should have some means to utilize SoC dtsi files and compose a single system devicetree that represents the boards and the connections between the various Soc's (in this case via SPI).

galak, 2021-03-18:
FYI @mbolivar-nordic, @carlescufi

carlescufi, 2021-04-19:
Completely agree with the premise. This of course should also include shared memory if any.

mbolivar-nordic, 2022-12-08:
Having become more familiar with the system DT specification, I think it would be challenging to do this. There doesn't seem to be a clean way to compose system devicetrees. The main challenges are that CPU clusters, as top level nodes, share a namespace. So if we want to be able to compose a system devicetree by `#include` of other SoC level system devicetrees, there is the potential for two CPU clusters with the same name to conflict. A similar problem exists in the /domains node.

I think this issue should be closed. As I have already communicated internally within the zephyr project efforts for this, I believe this is a non-goal for an initial effort.