devicetree-org/lopper #37: Forward lopper options to assist

etoillon, 2021-04-16:
Hello,

I'm using lopper assist to create files. The output directory is set by calling the lopper and the filename is set by config.dts. In order to have consistent output directory I would like to forward the output argument from lopper to assist as it does for verbose option.
I currently implemented it on my local repo

Thank you

zeddii, 2021-04-18:
Definitely a valid use case, and is something I've thought about.

If you have a starting patch to share, I'm definitely interested in seeing it (and then merging or completing, etc).

etoillon, 2021-04-19:
Hello,
Thank you for your quick answer
I attached a patch created from current master
[outdir_to_assist.zip](https://github.com/devicetree-org/lopper/files/6334702/outdir_to_assist.zip)

zeddii, 2021-04-21:
The patch looks fine to me.

Is there any chance you could regenerate it with your Signed-off-by: in the body of the patch. Alternatively, you could paste it here in the issue and I can copy it into the patch as I apply it.

We just want to make sure the attribution is correct and that we've handled it properly.

etoillon, 2021-04-23:
`From d3b6571079cfb7eaeaecb836811f1196e62cc5da Mon Sep 17 00:00:00 2001
From: Emmanuel Toillon <etoillon@kalray.eu>
Date: Mon, 19 Apr 2021 10:48:03 +0200
Subject: [PATCH] Forward output option to assist

---
 lopper.py | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/lopper.py b/lopper.py
index 35576bb..dee351d 100755
--- a/lopper.py
+++ b/lopper.py
@@ -1285,7 +1285,7 @@ class LopperSDT:
             if cb_funcs:
                 for cb_func in cb_funcs:
                     try:
-                        if not cb_func( cb_node, self, { 'verbose' : self.verbose, 'args': cb_opts } ):
+                        if not cb_func( cb_node, self, { 'verbose' : self.verbose, 'outdir' : self.outdir, 'args': cb_opts } ):
                             print( "[WARNING]: the assist returned false, check for errors ..." )
                     except Exception as e:
                         print( "[WARNING]: assist %s failed: %s" % (cb_func,e) )
-- 
2.17.1
`

etoillon, 2021-04-23:
Here the patch

```
From d3b6571079cfb7eaeaecb836811f1196e62cc5da Mon Sep 17 00:00:00 2001
From: Emmanuel Toillon <etoillon@kalray.eu>
Date: Mon, 19 Apr 2021 10:48:03 +0200
Subject: [PATCH] Forward output option to assist

---
 lopper.py | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/lopper.py b/lopper.py
index 35576bb..dee351d 100755
--- a/lopper.py
+++ b/lopper.py
@@ -1285,7 +1285,7 @@ class LopperSDT:
             if cb_funcs:
                 for cb_func in cb_funcs:
                     try:
-                        if not cb_func( cb_node, self, { 'verbose' : self.verbose, 'args': cb_opts } ):
+                        if not cb_func( cb_node, self, { 'verbose' : self.verbose, 'outdir' : self.outdir, 'args': cb_opts } ):
                             print( "[WARNING]: the assist returned false, check for errors ..." )
                     except Exception as e:
                         print( "[WARNING]: assist %s failed: %s" % (cb_func,e) )
-- 
2.17.1

```

zeddii, 2021-04-28:
Thanks for the update. The change is now in the tree.

closed 2021-04-28 by zeddii