devicetree-org/lopper #128: Incorrect usage of unit addresses in root nodes

mbolivar-nordic, 2022-11-24:
From the devicetree specification, section 2.2.2 Node Names:

> The
> unit-address must match the first
> address specified in the *reg* property of the node. If the node has no
> *reg* property, the *@unit-address* must be omitted and the
> *node-name* alone differentiates the node from other nodes at the same
> level in the tree.

From this we can also infer a derived requirement that any node with a unit address must have a parent node with `#address-cells` set, and the `reg` property must be consistent with it.

The system devicetree specification and associated examples violates this requirement in several places:

1. system-device-tree.dts

 https://github.com/devicetree-org/lopper/blob/da29be7393b1b3237c3a566b42a5dd5f81cf391c/specification/system-device-tree.dts 

This has a root node with `#address-cells = <2>`, so the `indirect-bus` and `cpus,cluster` child nodes should either 1. not have unit addresses or 2. have `reg` properties with 2 cells for the address. This is also true of multiple `simple-bus` nodes which are children of the root node.

2. specification.md 

https://github.com/devicetree-org/lopper/blob/da29be7393b1b3237c3a566b42a5dd5f81cf391c/specification/specification.md

Same issue: various nodes have unit addresses but no `reg` property. Additionally, the context makes it unclear whether the parent node has `#address-cells` set.

zeddii, 2022-11-24:
The captured reference tree needs some updating, it has been on the TODO list for a while. There are some other sample device trees floating around, but they are vendor specific, so I haven't wanted to directly use them.

We need to get @sstabellini to comment on this (hopefully tagging him here works)

sstabellini, 2022-11-28:
> From the devicetree specification, section 2.2.2 Node Names:
>
>
> From this we can also infer a derived requirement that any node with a unit address must have a parent node with `#address-cells` set, [...]

For `cpus,cluster`: there is no `reg` property so yes, there should be no `@address` in the cpu cluster name. They can be simply called `cpus-cluster0`, `cpus-cluster1`, etc.
Same for `indirect-bus`: there is no `reg` property so there should be no `@address` in the name.
Same for `simple-bus`: there is no `reg` property so there should be no `@address` in the name.

 
> 2. specification.md
>
>
> https://github.com/devicetree-org/lopper/blob/da29be7393b1b3237c3a566b42a5dd5f81cf391c/specification/specification.md
>
> Same issue: various nodes have unit addresses but no [...]

Yes they should be fixed as well. I'd be happy to take a pull request. Note that cpus actually have `reg` so it is OK to have `cpu@0`, `cpu@1`, etc.

closed 2022-12-01 by zeddii