devicetree-org/lopper #130: Purpose of secure-bus?

mbolivar-nordic, 2022-11-24:
When I was last working on system devicetree, we were discussing how to represent address maps for Arm v8-M CPUs with TrustZone support. In this case, the same IP blocks often effectively have different addresses depending on the security state of the CPU at the time of access.

It looks like secure-bus might be meant as a solution to that. Is that so? If so, I think it's not needed at least for Zephyr and it actually complicates things for us. I think having as many `address-map` properties as needed, with identical semantics (and therefore no differences in the places where the number of address and size cells are set; see https://github.com/devicetree-org/lopper/issues/129 for more on this) would be better.

In effect this is similar to the way the `pinctrl-<N>` properties work for pin control: any node with pin control support can have `pinctrl-0`, `pinctrl-1`, ... properties, as many as needed, and with identical semantics. Each property represents a different pin selection depending on the state of the IP block modeled by the node.

Similarly, as many address map properties as needed, each of which reflects some state of the cpus cluster, would seem to work well for us handling Arm v8-M in zephyr, at least for now. (This opens up the possibility of having an `address-map` or `address-map-0` property, then `address-map-1`, `address-map-2`, ... as well, but that is probably a separate issue.)

zeddii, 2022-11-24:
@sstabellini

sstabellini, 2022-11-28:
> When I was last working on system devicetree, we were discussing how to represent address maps for Arm v8-M CPUs with TrustZone support. In this case, the same IP blocks often effectively have [...]

That's great to hear and it looks like a good opportunity to simplify the specification. Would you be up for sending a pull request to remove `secure-bus` entirely?

mbolivar-nordic, 2022-11-29:
> Would you be up for sending a pull request to remove `secure-bus` entirely?

Certainly

closed 2022-12-01 by zeddii