devicetree-org/lopper #137: Domain-specific access flags are undefined

mbolivar-nordic, 2022-12-08:
This statement appears to be incorrect:

> Access flags are domain specific and have default values defined in
> the System Device Tree specification for each domain type. Different
> domains types have different compatible strings, in addition to
> "openamp,domain".

https://github.com/devicetree-org/lopper/blob/0da7cbd70bc2c3417e5e909bdd90eea2a0a0c912/specification/source/chapter3-execution-domains.rst

I don't see any place where these are defined.

mbolivar-nordic, 2022-12-08:
@sstabellini

sstabellini, 2022-12-09:
All three, but especially the flags in the *access* property of a domain. We have a few of those access flags defined for "xilinx,subsystem-v1" which are a type of domains supported by Xilinx firmware. They are used internally and not defined in this repository today, but we can certainly make them public.

The idea is that a Xen hypervisor domain is going to have different access flags compared to a xilinx,subsystem domain. Xen is using the IOMMU for isolation and Xilinx subsystems use the XPPU/XMPU which works differently and exposes different properties. These are xilinx,subsystem access flags:

  timeshare: bool
    type: boolean
    definition: bit 0
    description:
	  Time sharing or unique sharing. It sets the "Usage Policy" in PLM
	  CDO commands.

  allow-secure:
    type: boolean
    definition: bit 2
    description:
      If true, the device is allowed to generate secure accesses
      only applicable to masters. allow-secure is "Security Policy" in
	  PLM CDO commands.

  read-only:
    type: boolean
    definition: bit 4
    description:
	  If true, mapping is read-only, otherwise read-write.
	  It configures the "Write Policy" in PLM CDO commands.

  requested:
    type: boolean
    definition: bit 6
    description:
      Requested at boot.

  requested-secure:
    type: boolean
    definition: bit 36
    description:
      The device is secure or non-secure at boot.
	  Requested need to be set to true to use this flag.

  coherent: bool
    type: boolean
    definition: bit 37
    description:
      Routed via cache coherent interconnect.
	  Requested need to be set to true to use this flag.

  virtualized: bool
    type: boolean
    definition: bit 38
    description:
      Requested to be routed via SMMU.
	  Requested need to be set to true to use this flag.

  qos:
    type: <u32>
    definition: values [0-100] expressed using bits [64-95]
    description:
      Device specific QoS setting.
	  Requested need to be set to true to use this flag.

mbolivar-nordic, 2022-12-10:
> They are used internally and not defined in this repository today, but we can certainly make them public.

I think from the perspective of the system DT spec, it might make sense to adjust the language to say that it depends on the binding for the domain, and just leave it vague. Not sure if we want to start including vendor-specific bindings in the spec at this point in time; what do you think?

sstabellini, 2022-12-12:
I completely agree.