devicetree-org/lopper #144: How to add lable for existing node?

adally, 2022-12-27:
The format is: "path":"property":"replacement" to modify property. My original node is as below:
```
                cpu@0 {
                        device_type = "cpu";
                        compatible = "arm,armv8";
                        reg = <0x00 0x00>;
                        enable-method = "psci";
                };
```
And I want to change it to
```
                CPU0:cpu@0 {
                        device_type = "cpu";
                        compatible = "arm,armv8";
                        reg = <0x00 0x00>;
                        enable-method = "psci";
                };
```
How to add it?

When the existing device has some phandle and add a node with label, it will cause duplicated phandle issues. I used below operation to add a node.

 ```
       lop_2 {
            // add node
            compatible = "system-device-tree-v1,lop,add";
            node_src = "idle-states";
            node_dest = "/cpus/idle-states";
            idle-states {
                entry-method = "arm,psci";

                CPU_SLEEP_0: cpu-sleep-0 {
                    compatible = "arm,idle-state";
                    local-timer-stop;
                    arm,psci-suspend-param = <0x0010000>;
                    entry-latency-us = <40>;
                    exit-latency-us = <100>;
                    min-residency-us = <150>;
                };

                CLUSTER_SLEEP_0: cluster-sleep-0 {
                    compatible = "arm,idle-state";
                    local-timer-stop;
                    arm,psci-suspend-param = <0x1010000>;
                    entry-latency-us = <500>;
                    exit-latency-us = <1000>;
                    min-residency-us = <2500>;
                };
            };
        };

```
And I used dtc to covert the dtb file to dts file, it has below errors

```
$ dtc -I dtb fvp-base-revc-xen.dtb -o fvp-base-revc-xen.dts
fvp-base-revc-xen.dts: ERROR (explicit_phandles): /cpus/idle-states/cluster-sleep-0: duplicated phandle 0x2 (seen before at /bus@8000000/motherboard-bus@8000000/iofpga-bus@300000000/sysreg@10000)
fvp-base-revc-xen.dts: ERROR (explicit_phandles): /cpus/cpu0: duplicated phandle 0x3 (seen before at /refclk32khz)
fvp-base-revc-xen.dts: ERROR (explicit_phandles): /interrupt-controller@2f000000: duplicated phandle 0x1 (seen before at /cpus/idle-states/cpu-sleep-0)
ERROR: Input tree has errors, aborting (use -f to force output)
```

zeddii, 2022-12-28:
I wanted to follow up to let you know that I've seen this, but am out of the office until mid first week of January, so can't dig into this much before then.

Just so I'm clear, you are seeing two different problems:

1) writing a lop to add a label to a node
2) adding a node with phandles

As for #1, there is a label notation that needs be used, it may have issues. I need to run a test and will follow up.

As for #2, it is a bug. I'm not sure why it is generating a phandle and then not ensuring that it is globally unique in the target tree. I'll have to setup a test for this.

If you have sample input files, and lop files, can you attach them to the bug ? It will make reproducing the problem much faster on my end.

adally, 2022-12-29:
For #1 and #2, it has the same input files. [input_dts.txt](https://github.com/devicetree-org/lopper/files/10317217/input_dts.txt)
For #2, below is its lopper operation
```
     lop_2 {
           // add node
           compatible = "system-device-tree-v1,lop,add";
           node_src = "idle-states";
           node_dest = "/cpus/idle-states";
           idle-states {
               entry-method = "arm,psci";

               CPU_SLEEP_0: cpu-sleep-0 {
                   compatible = "arm,idle-state";
                   local-timer-stop;
                   arm,psci-suspend-param = <0x0010000>;
                   entry-latency-us = <40>;
                   exit-latency-us = <100>;
                   min-residency-us = <150>;
               };

               CLUSTER_SLEEP_0: cluster-sleep-0 {
                   compatible = "arm,idle-state";
                   local-timer-stop;
                   arm,psci-suspend-param = <0x1010000>;
                   entry-latency-us = <500>;
                   exit-latency-us = <1000>;
                   min-residency-us = <2500>;
               };
           };
       };
```

adally, 2023-01-10:
FYI, I am not blocked by this now. It's ok in my use case without adding an alias for CPU nodes and /cpus/idle-states.

zeddii, 2026-09-08:
Closing this, as you noted you were no longer blocked and found a way forward without adding aliases for the CPU nodes or `/cpus/idle-states`.

Apologies for leaving it open this long after saying I'd pick it up. If labelling existing nodes becomes a problem again, a fresh issue against a current release is the better starting point.

closed 2026-09-08 by zeddii