devicetree-org/lopper #158: How to use lopper to process multiple dtbs(with different compatibility)?

mengtanhzc, 2023-02-18:
Hello dear Lopper community, I want to try out lopper recently and have come across a problem.

I have some dtb files that are used on various boards, those dtb have different compatible and nodes, looks like:

```
// board-1.dts
/dts-v1/;
/ {
	compatible = "type1";

	...

	serial {
		...
	};
	...

   };
```

```
// board-2.dts
/dts-v1/;
/ {
	compatible = "type2";

	...

	serial {
		...
	};
	...

   };
```

I want to add different lops for them.
lop1.dts for board-1.dts, modify board-1.dts and extract some nodes to output-1.dtb.
lop2.dts for board-2.dts, modify board-2.dts and extract some nodes to output-2.dtb.

I'm going to handle them through a script, and generate the corresponding output-xx.dtb.

But if lop1 is used for board-2.dts, it may generate the wrong output1.dtb.
How can I do a compatibility check in lop1.dts? If compatible is not type1, is it possible not to execute the lopper operations that are defined in lops1.dts?
Thanks a lot!

zeddii, 2023-02-20:
There are several different ways to do this, some a bit more awkward than others.

You could do all the operations in a single 'code' lop, or you could also do it in an assist. But assuming you want to write a little python as possible, and would like to use the fundamental lop operations, that is where the bit of trickery comes into play.

If you are able to use the latest lopper, I just pushed a commit to master-next that makes things a little bit more streamlined.

You can see the details in the latest commit on master-next.

Fundamentally, you want to use the "select" lop to match on the compatible string of the dts files.

Once you've matched the root node, you can write code lops to check whether __selected__ is set (or not) and do conditional operations based on that. Without the change I referenced above, you'd need to always use code blocks to check __selected__ or another property that may be been set on the tree. With the commit I referenced above, you can refer to the result of the select lop and conditionally execute any type of lop based on the result.

Here's an example lop file that shows some of what I described above:

/dts-v1/;                                                                                                                                                                                                                                               
                                                                                                                                                                                                                                                        
/ {                                                                                                                                                                                                                                                     
    compatible = "system-device-tree-v1";                                                                                                                                                                                                               
    lops {                                                                                                                                                                                                                                              
            lop_1: lop_1 {                                                                                                                                                                                                                              
                  compatible = "system-device-tree-v1,lop,select-v1";                                                                                                                                                                                   
                  // clear any old selections                                                                                                                                                                                                           
                  select_1;                                                                                                                                                                                                                             
                  select_2 = "/:compatible:.*xlnx,versal-vc-p-a2197-00-revA.*";                                                                                                                                                                         
            };                                                                                                                                                                                                                                          
            lop_1_1: lop_1_1 {                                                                                                                                                                                                                          
                  compatible = "system-device-tree-v1,lop,code-v1";                                                                                                                                                                                     
                  code = "                                                                                                                                                                                                                              
                          tree.process = False                                                                                                                                                                                                          
                          if __selected__:                                                                                                                                                                                                              
                              print( 'Compatible dts (type1) found: %s' % node )                                                                                                                                                                        
[... cut ...]
            };                                                                                                                                                                                                                                          
    };                                                                                                                                                                                                                                                  
};

mengtanhzc, 2023-02-20:
> 

Many thanks! This looks great, I will try it based on master-next.

mengtanhzc, 2023-02-20:
> Fundamentally, you want to use the "select" lop to match on the compatible string of the dts files.
>
> Once you've matched the root node, you can write code lops to check whether **selected** is set [...]

Many thanks! This looks great, I will try it based on master-next.

closed 2023-02-24 by mengtanhzc