devicetree-org/lopper #163: Why do domains have a mandatory `id`?

mbolivar-nordic, 2023-03-02:
As of 4fc81d45e35cd4cf3ff19d2a837c6f15c1afbf4b, the "openamp,domain-v1" binding says `id` is required.

Why?

Why do we need an `id` at all, given that we can already refer to a
domain explicitly by its phandle, etc.?

At least within Zephyr, I cannot think of a reason why we would use
this property, so I think it should at least be made optional.

sstabellini, 2023-03-07:
Each Xen domain has an "id" and it is called "domid". If it is not specified, it would be automatically generated by Xen. Dom0 (domid == 0 is special so we need a way to detect if a domain has domid == 0 or domid != 0. PLM, which is the Xilinx firmware that handles AMP domains at the firmware level, also requires ids to identify domains (PLM calls them subsystems).

So, the reason why "id" is required is that certain types of domains are required to specify an "id". Could we find a way to make "id" optional by default but required depending on the compatible string? Maybe, but for simplicity we didn't go down that route.

Given that it is easy to add an "id" even when not necessary, and given the additional complexity of having to handle required vs. optional "id" cases, I don't know if it is worth making changes to the spec in that regard. 

Suggestions are welcome.

mbolivar-nordic, 2023-03-07:
> Given that it is easy to add an "id" even when not necessary

I disagree with this "easy" here -- in effect, this statement forces every potential Zephyr user of system devicetree to specify a useless value for every domain they configure. This will cause support burden and confusion.

> Suggestions are welcome.

This seems very Xen specific. It seems like a `xen,domain-id` property would be more appropriate.

sstabellini, 2023-03-07:
The id is certainly somewhat specific to the domain type. At this point, I don't know how many domain types need "id" and how many don't. Certainly, the presence of id could be made dependent on the domain type. Typically in device tree we try to reuse the same property name for the same thing even from different vendors, so I think it should be called "id" instead of "xen,domain-id". That way, it will be common across all domain types that require it.

If you are willing to make the changes to the specification to describe that:
- xilinx,subsystem* domains require "id"
- xen,domain* domains require "id"

Then I think we can make "id" optional for others.

I was not looking forward to maintaining a list of domains type that require "id" and a list of domain types that do not. That is way it was not done before.