devicetree-org/lopper #164: Shared access via `access` property

mbolivar-nordic, 2023-03-02:
As of 4fc81d45e35cd4cf3ff19d2a837c6f15c1afbf4b, the `access` property in the
"openamp,domain-v1" binding has the following in its specification:

```
   Description A list of devices the domain shall have exclusive access to,
               using bus firewalls or other similar technologies.
```

I can imagine scenarios where a precisely specified subset of the
execution domains in a system, each with a different CPU, needs access
to a resource, not just the default domain, a single domain, or a set
of domains that share a CPU.

Consider a scenario where an application processor needs to share a
peripheral at runtime with a helper core running an RTOS, with mutual
exclusion established at runtime via out-of-band means like an RPMsg
channel. This has 2 cpu clusters, each with their own domain, that
both need access to the peripheral. But absent other mutual exclusion,
in the system we want to prevent access from any other cluster, etc.

There doesn't seem to be a way to describe this in the current binding.

Am I missing something?

sstabellini, 2023-03-07:
We actually talked about resource sharing at length. If a resource is shared among multiple domains, it should be present in the "access" list of all the domains that are sharing the resource. Maybe it should be clarified in the specification.

mbolivar-nordic, 2023-03-07:
> Maybe it should be clarified in the specification.

It sounds like you're saying the spec should say:

> shall have access to

instead of 

> shall have exclusive access to

right?

sstabellini, 2023-03-07:
Yes