devicetree-org/lopper #170: Missing `reg` semantics for child nodes of indirect buses

mbolivar-nordic, 2023-03-02:
As of 4fc81d45e35cd4cf3ff19d2a837c6f15c1afbf4b, the spec seems to
leave the semantics here undefined:

```dts
/ {
	bus {
		compatible = "indirect-bus";
		clock {
			#clock-cells = <0>;
			compatible = "fixed-clock";
			clock-frequency = <32768>;
		};
	};
};
```

If we map in `bus`, do we always map in `clock`?

The current semantics for how nodes under an indirect-bus are included
or excluded rely on each node's `reg` property, which doesn't cover
this case.

sstabellini, 2023-03-07:
You have a good point, we should clarify this. I think if bus is mapped, then also clock is mapped.

mbolivar-nordic, 2023-03-21:
@sstabellini I thought about this more and I have another clarifying question: is this recursive?

For example, if `clock` has a child node without a `reg` property, is that also mapped?

sstabellini, 2023-03-22:
Yes, also the child of clock should be mapped