devicetree-org/lopper #409: A pre-defined lops directory to search lops files in lopper

onkarharsh, 2024-11-18:
Like assists implementation, where we have some pre-defined directories to check the assists files’ presence (e.g. lopper_directory, lopper_directory/assists and options passed with -A), we should have some pre-defined paths for lops files as well.

At present, we need to pass the full path/relative path of the lops file which is cumbersome and looks very odd in otherwise a sophisticated lopper command. Because we need full paths of lops file, we were forced to use lopper as a module in addition to its usage as a binary in our python scripts.

Adding pre-defined directory paths will make our life a little simpler.

Expected flow:
lopper -f --enhanced -O . -i lop-a72-imux.dts <sdt path> <out dts path> -- gen_domain_dts psv_cortexa72_0 linux_dt

Above command should first search for absolute path of lop-a72-imux.dts, if it doesn't exist, then it should search inside lopper/lops and then inside lopper directory. In addition, it should also have options to search inside user defined paths, similar to -A option for assists.

zeddii, 2024-11-20:
A follow up question to this. The -A option for assists is expected to be absolute, in that only the name of the assist (as passed on the command line is added to it). 

Would a similar behaviour be useful in a more generalized search path ?

I can definitely add the automatic addition of "lops" underneath the lopper executable (similar to assists), but I wanted to be sure that absolutely user supplied directories would be helpful.

I think they do need to be absolute, since if they are relative .. the question quickly becomes "relative to what" ?

onkarharsh, 2024-11-21:
Hi @zeddii ,

I agree with you, Having user defined absolute paths (similar to what assists have) makes sense.
User can always pass the individual lops file path with -i option where relative paths are already taken care of. So, I believe all the bases are covered with the user defined absolute paths to pass a different lops directory.

zeddii, 2024-11-21:
I'll also add the ability to set both assists paths and lop paths via environment variable. Just as another way to keep the command line simple, but have search paths specified.

zeddii, 2024-11-22:
https://github.com/devicetree-org/lopper/commit/7940ebfdb65c64d4cf78925e013e3727ae62c587

This has passed sanity and is now in master. If it has issues, or doesn't meet the need, let me know and I'll revisit.

closed 2024-11-22 by zeddii