devicetree-org/lopper #438: readme imporvements

pkp124, 2025-01-07:
Hello,

Suggestion to improve readme of this project, to help newbies like me. Things like
1. Prerequisities
2. Installation
3. Simple usage
4. Advanced options descriptions

These would be very helpful :)

Thanks

zeddii, 2025-01-10:
Some of this is covered, but I'm currently updating the READMEs to document some new options and the release schedule/process.

So I'll keep these in mind and make sure they are either reorganized to make them more visible or add the missing content.

zeddii, 2026-09-08:
Coming back to this a lot later than I should have, but the documentation has moved on since you filed it and I think your four points are reasonably covered now.

Against your list:

- **Prerequisites and installation** — `README.md` has a config/setup section covering the git flow, a venv-based flow, and installing from pypi.
- **Simple usage** — the "Sample run" section, with "Limitations" alongside it so the boundaries are visible.
- **Advanced options** — the full option list is in the README, and mirrored from lopper's own `--help` so the two don't drift.

Beyond the README, `docs/` has grown a fair bit since you asked: a developer guide covering how to write lops and assists, `sdt-from-linux.md` for building a system device tree from upstream sources, and architecture notes for a few of the larger assists.

Your ask nudged some of that along, so thanks for taking the time to file it as a newcomer. That perspective is genuinely harder to get from people who already know the tool.

Closing on that basis. If there's still something specific that tripped you up, a focused issue naming the gap is much easier to act on than a general documentation request, so please do open one.

closed 2026-09-08 by zeddii