devicetree-org/lopper #544: xlnx_overlay_pl_dt ignoring "domain" overlay

traavnik, 2025-04-17:
Hi, first of all sorry for any terminology mix-ups. I'm still very much a beginner in all this DT jazz. I'm also unable to determine what's work in progress and what's considered done but am only using main branch.

I'm trying to extract *pl.dtsi* for an AMD MPSoC design and have gone through the SDT workflow, additionally providing *pl-overlay.dtsi* using `-include_dts` flag. This made `#include "pl-overlay.dtsi"` appear at the end of the *system-top.dts* file and *pl-overlay.dtsi* copied over.

I tried using both *xlnx_overlay_dt* and *xlnx_overlay_pl_dt* for these and found that the latter didn't pull in any of the domain specific overlays. I guess this is either a bug or I'm unable to use the assist properly.

Example:

*pl-overlay.dtsi* contains:
```
&foo_v1 {
	compatible = "generic-uio";
	status = "okay";
};t
```

*xlnx_overlay_dt* outputs *pl.dtsi* containing:
```
&amba{
	foo_v1: foo_v1_0@a0000000 {
		compatible = "generic-uio";
		interrupts = <0x0 0x5a 0x4>;
		xlnx,s-axi-data-width = <0x20>;
		xlnx,rable = <0x0>;
		interrupt-parent = <&gic>;
		xlnx,ip-name = "foo_v1_0";
		reg = <0x0 0xa0000000 0x0 0x10000>;
		clocks = <&zynqmp_clk 0x47>;
		xlnx,edk-iptype = "PERIPHERAL";
		status = "okay";
		clock-names = "s_axi_aclk";
		xlnx,s-axi-addr-width = <0x4>;
		interrupt-names = "axi_interrupt";
		xlnx,name = "foo_v1";
	};
};
```

while *xlnx_overlay_pl_dt*  outputs *pl.dtsi* containing (note unchanged *compatible* field):
```
&amba {
	foo_v1: foo_v1_0@a0000000 {
		compatible = "xlnx,foo-v1-0-1.0";
		interrupts = < 0 90 4 >;
		xlnx,s-axi-data-width = <32>;
		xlnx,rable = <0>;
		interrupt-parent = <&gic>;
		xlnx,ip-name = "foo_v1_0";
		reg = <0x0 0xa0000000 0x0 0x10000>;
		clocks = <&zynqmp_clk 71>;
		xlnx,edk-iptype = "PERIPHERAL";
		status = "okay";
		clock-names = "s_axi_aclk";
		xlnx,s-axi-addr-width = <4>;
		interrupt-names = "axi_interrupt";
		xlnx,name = "foo_v1";
	};
};
```

zeddii, 2025-04-21:
Can you elaborate on how you are running lopper ? Are you running it standalone on the command line, or is it as part of some of the AMD/Xilinx tools (like genmachineconf) that wrap lopper ?

That should help me tag the right people on this issue and get their thoughts.

traavnik, 2025-04-22:
Sure, sorry! I'm using Lopper directly (commit 5a78f6d), installed following README.md instructions.
I have edited the original post with the exported flags and the two commands used to generate outputs.

zeddii, 2026-09-08:
Apologies for the long silence here — you supplied everything I asked for and then this sat.

`xlnx_overlay_pl_dt` has since gained the ability to apply user overlay content, which is what was missing when you filed this:

- `d5793595` — auto-discovers `*.dtso` files co-located with `system-top.dts` and registers them through the same compilation path as `-i`, then emits them as `&label` fragments in `pl.dtsi`
- `0675e616` — adds `--exclude-overlays` to opt files out of that discovery

Your overlay arrived as a `#include`'d `.dtsi` rather than a co-located `.dtso`, so I can't say with certainty that your exact invocation now works. But the capability you were after is there, via either a co-located `.dtso` or an explicit `-i`.

If you are still working with this, could you retry on a current release? If it behaves, I will close this out. If it still ignores the overlay, the reproduction you already posted is enough for me to dig in properly this time.

traavnik, 2026-09-28:
Hi, I've quickly tested the two assists again and it seems that `xlnx_overlay_pl_dt` now does apply the overlay.
`xlnx_overlay_dt` doesn't seem to do so anymore tho. I'm not sure if this is by design or not. I also had to add the `<PL DTSI>` part to the `xlnx_overlay_dt` call which was correctly suggested by the script's  *Usage* message but is not explained in [README](https://github.com/devicetree-org/lopper/blob/b911af37a655accdd9be87f989282941a04c15f5/lopper/assists/README).

As far as I'm concerned this issue can be closed. The problem from the title has apparently been solved and I don't really care about the *old* workflow.

Thanks!