devicetree-org/lopper #546: Creation of .pp files inside lops directory by default for every lops processing

onkarharsh, 2025-04-25:
At present, whenever lopper processes any lops file, it tries to create the intermediary .pp files within the lops folder. When we try to ship lopper as a tool, more often we don't have the write permissions inside the lops folder. At present, we are forced to use -O option to redirect the .pp generation to a writable area. I propose to change the default .pp file location from the lops folder to the present working directory. Passing an extra -O option all the time seems cumbersome for the users.

zeddii, 2025-04-25:
I agree that going to the directory of file being processed isn't ideal, but the current working directory was tried early on in development and discarded as it had similar issues (it isn't always writeable).

I have a patch that moves the preprocessing to a temporary directory.

That should meet the need, do you agree ?

onkarharsh, 2025-04-26:
Thanks @zeddii . Moving to a tmp directory is a better approach, that would help.

zeddii, 2025-04-29:
I have this mostly working now, I'm chasing a corer case with lop files that are still having a .pp file written to a non-temporary directory. I should have a patch available for test shortly.

zeddii, 2025-05-05:
https://github.com/devicetree-org/lopper/commit/bdbe71ed8d14329c3267b67db6934dcdb7da8f0e

closed 2025-05-05 by zeddii