devicetree-org/lopper #551: Assists and Lops documentation

onkarharsh, 2025-05-16:
Hi @zeddii , 
We are working towards documenting all our assists and lops that we use during our embedded software stack build process. As part of that, we would like to understand where we can put these rst files. I believe it is time that we start creating enterprise specific assists and lops directories. Your thoughts?

CC: @sivadur @SamPovilus

onkarharsh, 2025-05-16:
Also, Is there a way to get the assists help/usage using lopper front end? something like : lopper baremetal_xparameters_xlnx --help

I see the usage prints in here https://github.com/devicetree-org/lopper/blob/master/lopper/assists/extract-xen.py#L31 , How to use this?

onkarharsh, 2025-05-19:
Thanks @zeddii for the response.
Yes, not mandating an input-output file for assist to get the help message would be useful. That helps.

zeddii, 2025-05-20:
https://github.com/devicetree-org/lopper/commit/551a67fd6c9f16252ad4b6a55e266a08d38d957c

I just need to let this soak for a few days to see if I missed any corner cases on the handling.

onkarharsh, 2025-05-21:
Thanks for the confirmation @zeddii . We will create a separate doc directory for AMD and leave the existing lops/assists as is. We will soon send the Pull Request for the same. Please feel free to close this Issue once your updates for assists' help is in.

zeddii, 2026-09-08:
Closing as requested. To summarise where this landed:

- Assists and lops stay in the common directories rather than being partitioned by contributor, so they keep serving as reference material.
- Getting help from an assist without supplying input and output files went in as `551a67fd`.
- Separate documentation directories are fine, and were the approach you settled on.

Since this was filed, the developer guide has also grown a section on the code lop execution environment — which names are in scope, how to reach an assist via `inherit`, and how to pass a value in per invocation.

closed 2026-09-08 by zeddii