devicetree-org/lopper #553: Minor query regarding no-libfdt use case

onkarharsh, 2025-05-19:
https://github.com/devicetree-org/lopper/blob/master/lopper/__main__.py#L234

There is a user argument --no-libfdt. Just curious to know under what circumstances user would like to exercise this option. I see that LopperDT class is not as enriched as LopperFDT. Any future plans?

zeddii, 2025-05-21:
When I was adding the support for the two dt input mechanisms, I found that it wasn't possible to assign that value a default and then be able to later override it based on the command line option.

That could have been an idiosyncrasy of the python version I was using at the time.

Due to the way the lopper modules are structured, there's some interplay and initialization order that makes it tricky to globally override that variable and have other imports work without circular references. That's how the implementation landed as it currently stands.

That being said, I can give the global Lopper variable a default as such:

 diff --git a/lopper/__init__.py b/lopper/__init__.py
index 4155db1..57ba468 100644
--- a/lopper/__init__.py
+++ b/lopper/__init__.py
@@ -19,6 +19,7 @@ import tempfile
 from collections import OrderedDict

 from lopper.fmt import LopperFmt
+from lopper.fdt import LopperFDT

 from lopper.tree import LopperNode, LopperTree, LopperTreePrinter, LopperProp
 import lopper.tree
@@ -34,6 +35,10 @@ except Exception as e:
     print( f"[WARNING]: cant load yaml, disabling support: {e}" )
     yaml_support = False

+# Default processing type
+Lopper = LopperFDT
+lopper.tree.Lopper = Lopper
+
 @contextlib.contextmanager
 def stdoutIO(stdout=None):
     old = sys.stdout

And everything still appears to work. I've added that to my local commits and will let it soak for a few days.

zeddii, 2026-09-08:
Following up on this rather late ... the default assignment I sketched did land, so the workaround you were carrying shouldn't be needed any more.

`lopper/__init__.py` now has:

```python
from lopper.fdt import LopperFDT
...
# Default processing type
Lopper = LopperFDT
lopper.tree.Lopper = Lopper
```

with the command-line override kept as `lopper_type(cls)`, which reassigns both the module global and `lopper.tree.Lopper` when `--no-libfdt` is passed. So the variable is defined at import time and still overridable, which was the part I wasn't sure was possible when you first raised it.

Confirmed against current master that the sphinx case works without `main()` being called:

```
from lopper import Lopper  ->  LopperFDT
lopper.tree.Lopper         ->  LopperFDT
lopper_lib imports clean
```

You should be able to drop the hardcoded `Lopper` from your local `__init__.py`.

On the original question about `--no-libfdt`: that stands as answered above, it exists for build systems without c-bindings to python, or where python-only execution is preferable. It's less exercised than the libfdt path because the non-libfdt parser is stricter about inputs that dtc accepts, and it becomes more viable as schema support grows in lopper directly. And in fact, the zephyr project doesn't seem to support the parser that it was using anymore, so it will need to be refreshed and updated to something different in the not to distant future.

Closing this out, but do reopen or file fresh if the API docs turn up anything else awkward.

closed 2026-09-08 by zeddii