devicetree-org/lopper #619: [ERROR]  [tree.py][WARNING]: Exception (tunes) raised by code block:

likewise, 2025-08-26:
How can I debug this issue?

This seems a piece of dynamically generated Python code to be executed which then raises an exception.

```
../sources/meta-xilinx/meta-xilinx-core/gen-machine-conf/gen-machine-conf --hw-description ../sdt -c ./conf/ 
NOTE: Starting bitbake server...
NOTE: Started PRServer with DBfile: /project-on-host/build/cache/prserv.sqlite3, Address: 127.0.0.1:37693, PID: 66303
NOTE: Unpacking /project-on-host/sdt to /project-on-host/build/hw-description/
[INFO] Using HW file: /project-on-host/build/hw-description/project-on-host/sdt/system-top.dts
WARNING: XSCT has been deprecated. It will still be available for several releases. In the future, it's recommended to start new projects with SDT workflow.
[INFO] Getting Platform info from HW file
[WARNING] The lopper 'lops' or 'embeddedsw configuration' files are missing, Trying to get recipe sysroot using bitbake
[WARNING] Unknown CPU xlnx,microblaze-11.0
[WARNING] Unknown CPU xlnx,microblaze-11.0
[INFO] Silentconfig project
[INFO] Generating configuration files
[INFO] Generating cortex-a53 Linux configuration [ None ]
[INFO] Generating cortex-a53 baremetal configuration for FSBL
[INFO] Generating microblaze baremetal configuration for ZynqMP PMU
[INFO] Generating microblaze processor tunes
[ERROR] 

[tree.py][WARNING]: Exception (tunes) raised by code block:
def __node_test_block():

                                   if __selected__:
                                       for n in __selected__:
                                           n.tunes['reorder'] = 'reorder'
                          
__nret = __node_test_block()
```

zeddii, 2025-08-26:
The exception in itself isn't an error (it is how the framework catches issues with the python in lopper operations (lops)), which is what is happening here.

It isn't that hard to debug, but the raw lopper command and inputs are needed, since we need to add some verbosity to the run and look at the files.

Looking at the thrown exception, in this case the lop is looking at nodes in the device tree and accessing a field called "tunes", the nodes that have been selected in the other lopper operations (__selected__) don't have that field, so we get the exception.

Since this works in other scenarios (that specific lop, not lopper itself), there's likely something wrong with the inputs which also likely stem from the WARNING lines in the log (But I know nothing about gen-machine-conf, so I can't say more than that).

If you can find the inputs and lopper command line, I can definitely have a look.

likewise, 2025-08-26:
I think I can solve the issue, and close it here. One question is; how can I more easily debug similar issues next time? I mainly found the bug by reading code.

I noticed I cannot insert print() into the lop files, correct?

zeddii, 2025-08-27:
Almost any core python can run inside of a lop code block (there are restrictions on some modules/commands, but that's not relevant here). So print is definitely available. You just can't use quotes, since it will cause a parse error. quotes need to be escaped, or just use single quotes (which is what I typically do).

You can also add increasing levels of verbosity to lopper with repeated -v arguments to get more information. 

Let me know if there's anything more I can do to help solve the issue!

zeddii, 2026-09-08:
Closing this out, since you found the cause and the debugging question is answered: `print()` works inside a code block, the constraint is on quoting rather than on the call itself.

Reopen if the `tunes` path bites again on a current release.

closed 2026-09-08 by zeddii